volumio/volumio-plugins #430: "No Results" error when using the YouTube plugin

adamj-codethink, 2020-04-20:
I have signed in to the YouTube plugin and allowed access, i have also set some music play lists to public, but the plugin keeps reporting "No Results" when i click any section of the plugin.

Is there anything else i need to enable, or places i could look for logs?

Thanks

balbuze, 2020-04-20:
Please, Youtube plugin is removed from official plugin because it does not work, so don't use it!
It willl be available when fixed.
Remove it to make search work with other sources.

adamj-codethink, 2020-04-20:
Thanks, it still appears under my Volumio "plugins" section, i will remove for now

balbuze, 2020-04-20:
You're right, it still appears. To be fxed. Thanks!

volumio, 2020-04-20:
Ah, I thought we removed it.. Thanks for reporting..

@balbuze can you please remove it from the gh-pages branch?

balbuze, 2020-04-20:
done...

adamj-codethink, 2020-04-20:
i will close this for now :)

closed 2020-04-20 by adamj-codethink