volumio/volumio-plugins #442: Plugin update glitches

arjenpdevries, 2020-05-16:
Just to raise attention to problems with updating the volspotconnect2 plugin to the most recent version, as reported in https://github.com/balbuze/volumio-plugins/issues/265

@ashthespy helped me resolve the immediate problem on my device by simply first uninstalling instead of using update - so the problem is not urgent, as long as people realize this might resolve the problem for them as well.

But, the diagnosis indicates that the plugin update is incomplete, and as a result old commands are issued but `package.json` and UI report the newly updated plugin.

ashthespy, 2020-05-16:
Thanks for the follow up! Just to add some specific info, it looks like some files are getting updated, while some are not. 
At first glance, `startconnect.sh` doesn't get updated. I peeked into the [`volspotconnect2.zip`](https://github.com/volumio/volumio-plugins/blob/gh-pages/plugins/volumio/armhf/music_service/volspotconnect2/volspotconnect2.zip) file to confirm that we weren't making some mistake during packaging. But for some reason, this file isn't getting overwritten when the plugin updates.. Maybe some race condition with `systemmd` as this is the file the service calls?

gvolt, 2020-06-12:
I encountered a similar issue. Recently I was refreshing the minidlna plugin (version 1.1.4, installed from the plugin manager) with my development version (which has a higher version number than the currently published version) by `volumio plugin refresh`.

The updated files were all correctly put into /data/plugins/miscellanea/minidlna, but it takes a reboot or executing `volumio vrestart` to get them processed by Volumio; stopping and starting the updated plugin does not suffice. It seems as if the pre-update files are cached somehow.

During the update process I watched `sudo journalctl -f` but there have not been any error messages.