volumio/volumio-plugins #446: Feature Request: Spotify: list associated devices and allow playback on them

PartisanEntity, 2020-05-27:
In Spotify you can have devices associated with your Spotify account. When playing music you can select a device:
[image]

It would be great if one could see those devices in the Volumio Spotify plugin and choose to play music on those devices.

Similar to how Volumio will display Audio Outputs & Zones.

balbuze, 2020-05-27:
Install the plugin volspotconnect2 : Spotify connect...

PartisanEntity, 2020-05-27:
Thank you very much for the suggestion. I have tried that, but it simply turned the Volumio instance on which it was installed into a Spotify receiver.

I would however like to choose to pay a Spotify song on my Sonos speaker from within Spotify on Volumio