volumio/volumio-plugins #449: Latest touch screen plugin does not install properly on v.2777 volumio

jack2675, 2020-06-02:
hello,
The installation of latest touch screen plugin gives "error" or rather impossibility to install chromium and unmet dependencies on a fresh installation of volumio v2.777. Message on libraspberrypi0 are also given to say that it cannot be modified. The plugin installation finally ends but impossible to see the volumio UI once the plugin enabled.

Please notice the following information:
- I use RPI 3B and I reached to make it work with the previous revision of plugin 3 weeks ago in the same configuration by using volumio-2.773-2020-05-05-pi.img.
- After update with v2.777 it was still working
- Then I crashed my installation trying to install touch driver for 5" display and flashed a brand new SD card
- I restarted using v2.773 but now only the new plugin is proposed.... and then it does not work for me as before...
Please help if you can, thanks!

gvolt, 2020-06-02:
Reason is an expired certificate (just expired on 2020-05-30). This impedes the download of libraspberrypi0_0.0.1_all.deb, raspberrypi-bootloader_0.0.1_all.deb and raspberrypi-kernel_0.0.1_all.deb from repo.volumio.org atm.

You may have a look at the forum for further information:

- [https://community.volumio.org/t/raspberry-pi-7-touch-screen-plugin-not-working/38747/6](url)

- [https://community.volumio.org/t/berryboot-screen-raspberry-is-black/38822/3](url)

- [https://community.volumio.org/t/ssl-root-certifcate-validation-for-podcast-plugin-fixing-the-breakage-from-the-addtrust-external-ca-root-expiration/38828](url)

jack2675, 2020-06-02:
Thanks a lot gvolt. Is there a temporary workaround or should we have to wait renewal of certificates and a new release of the volumio plugin? Thanks!

gvolt, 2020-06-02:
The plugin does not need a new release to solve the issue. "Just" the certificate situation has to be cleared. If you do not want or cannot wait for that you can either try the workaround of [billy_pi](https://community.volumio.org/t/raspberry-pi-7-touch-screen-plugin-not-working/38747/6) which uses the --no-check-certificates flag on the related wget commands or the way [ashthespy proposed](https://community.volumio.org/t/ssl-root-certifcate-validation-for-podcast-plugin-fixing-the-breakage-from-the-addtrust-external-ca-root-expiration/38828/2). The latter certainly does require less manual action.

volumio, 2020-06-02:
Ok, fixed the certificate

closed 2020-06-02 by volumio

jack2675, 2020-06-03:
Thanks a lot for all these hints. I saw in one of them raised by you that issue from certificates is now solved! Many thanks!