volumio/volumio-plugins #451: Sound doesn't pass after Simple Equalizer installation

dmitrij9992905, 2020-06-13:
I install volumio equalizer plugin, and once I enable it, sound disappears.
Volumio works on Raspberry Pi4, using external sound card based on CM6206.
And if sound appears, I can't hear sound difference after equalizer changing. 
Also I tried to change sound device - nothing

balbuze, 2020-06-13:
Disable the plugin. Reboot. Then ,in volumio playback set your sound card and volume mixer (if any). Test with a track. 
Enable the plugin. DO NOT change anything in Volumio playback! Reboot and test.

bkuhls, 2020-07-02:
Same here:
https://community.volumio.org/t/volumio2-equalizer-plugin/6376/167
https://community.volumio.org/t/volumio-simple-eq-no-sound/39504