volumio/volumio-plugins #454: volspotconnect2 and caching

fuerst, 2020-06-23:
What is the reason to disable caching in  _/data/plugins/music_service/volspotconnect2/volspotify.toml_? Is it because the cache size can not be limited?

I activated it anyway and caching seems to work. The _files_ directory get's filled up with files as I play Spotify songs. Probably the cache needs to be moved to it's own size limited partition or volume so it will not fill up the whole SD card.

ashthespy, 2020-06-25:
Well, this is multifold: 
-- There is indeed currently no measures in place to limit the size of the cache.
-- If cache is set to `/tmp`, then its on RAM, so a large cache would impact system performance at some point. 
-- Directly writing onto the SD card is kept to a minimum to prevent SD card wear.

balbuze, 2020-06-25:
In earlier version of the plugin, I had a cache handling with auto flush when a define size was reached. Maybe we can pull it from dust?

fuerst, 2020-07-08:
Thanks for your answers! 

I think with durable SD cards like Samsung PRO Endurance available it would be okay to write directly to the SD card. Also using a big SD card rather than a small one would help because data will be written less often to the same cell. 

So if you agree I think we need:

1. Additional options in the UI which allows 
  - activating caching
  - set the filesystem path for cache files with a default in the volumio user's home directory
  - set the cache size
2. Use the script from b0a2bbd319c2632f087d9bd550fcd719ebaf3ca0 to monitor the cache size and remove oldest files if needed.

How does the the cleanup script get triggered - by using cron?

balbuze, 2020-07-08:
It uses a system d timer as far I remember