volumio/volumio-plugins #554: Tradfri-plugin loses connection to the gateway

JayJay-Master, 2021-03-10:
I installed the tradfri plugin, made all the settings and it works fine. After a while (some hours or over night) it will lose the connection to the tradfri gateway because it no longer works. Only after pressing disconnect, entering the security code again and reconnecting, it works again as it should.
Does anyone have an idea why this might be. The IP address of the tradfri gateway is unchanged all the time. 

System Information:
Version of Volumio: 2.873
Kernel: 4.19.118-v7+
Raspberry Pi 3 Model B Plus Rev 1.3 with HiFiBerry DAC Plus

Thanks!

F0gRex, 2021-06-06:
I did a lot of debugging because I had some issues as well.
Could you resolve your problem by now?

JayJay-Master, 2021-06-07:
No, I still have the problem. Only disconnection and reconnection solves it for a while.

F0gRex, 2021-06-16:
Hi,

When I have time I might try to fix this issue / provide a workaround.

I just could reproduce the error but I have no clue so far why it is happening because the rest of the plugin is still working and no error are displayed in the log.

Can u provide additional information?

Another issue is that when saving the switch-off delay it does not display the changes after reloading (the values get updated in the background and everything works fine).

Do you have this problem as well?
What the release version of your tradfri gateway?

F0gRex, 2021-06-16:
Update:

I am not really a professional in JS but this is what i found out:

The problem occurs every time when the plugin is reloaded.
The connection should get reestablished at line 137 in the index.js file.

Unfortunately the code is pretty messy sometimes but there is for sure a mistake in line 137 in my opinion.

It should be 
`.then(self.onTradfriGatewayDiscovered.bind(self, ''))`

instead of 
`.then(self.onTradfriGatewayDiscovered.bind(self))`

The reconnection should then work.

F0gRex, 2021-06-20:
@JayJay-Master Can u check if your issue is fixed too?

JayJay-Master, 2021-06-22:
@F0gRex
Hi Joel,

sorry for the late reply, but I was out a few  days.

I implemented the fix you provided and it looks good so far. Even after a restart, the Tradfri plugin automation works flawlessly. If I remember correctly, the problem occurred after a reboot, but sometimes suddenly after a few days. So I need a little more time to finally confirm the fix.

As for the other issue, I cannot confirm it. In my installation, the "Switch Off Delay" time is displayed even after entering the settings of the Tradfri plugin again. The time is displayed and it works fine.

Thanks for your efforts so far. I will give a status update for your fix .in a few days.

JayJay-Master, 2021-06-22:
The release version of the Tradfri plugin is 1.0.1.

JayJay-Master, 2021-06-22:
Now I just updated to version 1.0.2 with your fix.

F0gRex, 2021-06-22:
i messed up with my fix btw 
update 1.0.3 should fix it when pulled

F0gRex, 2021-06-22:
Strange, so if you set the switch off time to e.g. 10 seconds and reload the page, it still displays 10s ?

In my case it simply displays the default 60s (but in the background the new value 10s is memorized correctly)

JayJay-Master, 2021-06-22:
Hi,

I saw version 1.0.3 in Volumio, but 1.0.2 was installed.
Version 1.0.2 contains the following below mentiond change:

> F0gRex added a commit to F0gRex/volumio-plugins that referenced this issue 6 days ago
> @F0gRex
> [Fixed reconnection issue with the tradfri plugin](https://github.com/F0gRex/volumio-plugins/commit/5b2fe39d0bb625c8d41140f18086c2318d8ff2a9) 

With this version after reboot I loses the connenction again. 

So I changed the index.js on my local installion to add your first mentioned fix in **line 137** addionally:

```
135   135       if (identity && psk) {
136   136         tradfriLib.discoverGateway()
137          -    .then(self.onTradfriGatewayDiscovered.bind(self))
      137    +    .then(self.onTradfriGatewayDiscovered.bind(self, ''))
138          -    .catch(self.onTradfriConnectError.bind(self));
      138    +    .catch(self.onTradfriConnectError.bind(self, ''));
139   139       }
140   140
141   141       return libQ.resolve();
```
This works also after reboot.

JayJay-Master, 2021-06-22:
I will give an update on this fix in a few days.

JayJay-Master, 2021-06-22:
Regarding the 2nd problem with "switch-off delay":

It was set to 60 seconds. If I change it to 10 seconds, safe and go back to the plugin menu, it still shows 60 seconds but it turns off after 10 seconds. So it seems I have also the problem you described. 

Before I tested only if the value ist displayed when reentering the plugin settings, but obviously I never changed the default value. Sorry fore thhat

JayJay-Master, 2021-06-22:
Updated now to version 1.0.3.  I will give an update on this fix in a few days.

F0gRex, 2021-06-22:
To make things clear.

Update 1.0.2 is not a fix (it actually contains a bug more (reporting an error would not work as intended because of my changes)
This is due to the fact that i accidentally edited the wrong line.

But i fixed this issue (reversed my change and added the correct change) with version 1.0.3 but this change is not yet in the official repository.

The wrap things up: The only change you need to do is on line 137.
The change on line 138 is wrong but would only take effect when the plugin would throw an error.

Concerning the other issue with the switch-off delay.
I am more or less glad that you experienced the same issue (everything else would have been extremely strange)
I am currently quite busy but I will try to fix this issue in the upcoming weeks as well.

F0gRex, 2021-06-28:
update 1.0.3 is now officially released and your issue should be fixed

JayJay-Master, 2021-06-29:
Version 1.0.3 works fine. No more disconnection to Tradfri gateway. 
With respect to the dosconnection error issue can be closed. For the switch-off delay issue I keep the ticket open and leave it to F0gRex to continue this here or to open a separate ticket.

F0gRex, 2021-06-29:
I did already create issue #592 for addressing this problem.

Therefore this issue can be closed

JayJay-Master, 2021-06-30:
Issue closed. Thanks to F0Rex!

closed 2021-06-30 by JayJay-Master