volumio/volumio-plugins #579: Cannot add podcast episode to Favorites

teq0, 2021-05-17:
In the UI it gives you the option to save a particular podcast episode to Favorites, and when you do it comes back with a message saying "Added <podcast episode name> to Favorites". 

Except when you go to Favorites it's not there.

teq0, 2021-12-02:
I'm happy to have a go at fixing this if someone can point me in the right direction. I've scanned the code and it's not clear what needs to be changed i.e. it's not clear to me how adding favorites actually works.

teq0, 2022-01-08:
@ChrisPanda I've made some progress, I can get podcasts saved into Favourites OK, and under certain circumstances they will even play from Favourites, but there are a couple of fairly large issues that I can see.

Firstly, the code relies on `self.currentEpisodes`, which may not be populated if the user goes straight to Favourites after Volumio starts up, or could contain the episodes for a different podcast, if the user had recently browsed to a different podcast. Right now this will either kill Volumio with an error or play the wrong podcast.

I think the fix for this is to store the list of episodes under each podcast in `self.podcasts`, rather than a single "global" list of the last podcast loaded, and to look up episodes using both the podcast name and the episode number, and also to load the list from the RSS URL on demand so that it's not necessary for the user to have already browsed to the specific podcast.

Secondly, the URI scheme is based on integer indexes (e.g. `podcast/3/2`), which means that if a podcast is ever removed from the list then the URI for any existing favourites for podcasts that were _later_ in the list than the one that was removed will now have incorrect indexes. I think the only way to fix this is to change the URI scheme to use the name of the podcast as the second item, for example `podcast/The Daily/3`, similar to how folder paths are used in MPD favourites. It's doable but will require quite a lot of changes, and is complicated by the fact that BBC podcasts have their own special logic. 

Obviously if the Favourite refers to an episode of a podcast that has been removed then there's nothing we can do, since we won't have the RSS URL to read the episode list, so that should return an error to the user asking them to add the podcast again or delete the favourite.

I think it's still OK to use the integer index for the episode number in the URI, as these are supposed to be immutable.

What do you think? Can you see any simpler solutions?

I'm going to start playing with changing the URI scheme to see how much work there actually is.

teq0, 2022-01-09:
I've implemented the new URI scheme, with the name instead of index for the second element, and the "on demand" loading of podcast items, it works well, except that the core code calls `explodeUri` for every item in the Favourites list whenever you click on a single item. This means that if you have episodes from 10 different podcasts in your Favourites then the first time you click on _any_ favourite, podcast or not, it goes and fetches the RSS for every podcast. I added a cache timeout so that it doesn't fetch multiple times for the same podcast, but the user still gets the toaster popups about fetching the content for all of the podcasts.

It's not terrible but it's probably unexpected behaviour for the user.

ChrisPanda, 2022-01-10:
@teq0 

Hi, you don't need to change URI scheme and I already fix the problem and it will be pull request to Volumio and the explodeUri has a bug for a favorite function. In addition, I almost finish the podcast directory searching feature.

Thanks

teq0, 2022-01-10:
OK, looks like I wasted a few days of my time then. I assumed from your response on the forum that you were happy for me to work on this and that you were working on the search function.

Curious to see your solution, especially how it handles deleted podcasts while still using a simple integer index in the URI.