volumio/volumio-plugins #639: Plugin touch_display - Sometimes getDisplayNumber returns an empty string and displayNumber remains empty

danowar2k, 2021-11-01:
There is a variable displayNumber which seems to be only set in the getDisplayNumber function. This function is only called in "onStart".

The displayNumber variable is used in several other places, like a static variable which is expected to be set during "onStart".

But there seem to be instances when getDisplayNumber sets displayNumber to an empty string. Perhaps there are asychronous things happening where a condition is true so that the plugin believes now is the time to write the displayNumber or something.

Anyway, this leads to an empty string, which in turn leads to the plugin not finding the unix socket, because it looks for a socket called '/tmp/.x11-unix/X'.

Perhaps it would be better to always use the getDisplayNumber function instead of the displayNumber variable. And if you want static behaviour, check at the beginning of the function if displayNumber already has a non-empty string. And then only do the xinit service check if it doesn't.

gvolt, 2021-11-11:
@danowar2k 

Sorry for the delay, but here are my thoughts:

The plugin first determines the Xserver socket it uses. To do so, the function "getDisplayNumber" evaluates the result of the query of the status of the systemd service "volumio-kiosk.service". If the Xserver is already started, the status message contains among other things the name of the Xserver socket, typically (but not necessarily) "X0" where the "0" is the display number on which the Xserver is running.

Since it is impossible to predict when exactly the Xserver startup will be complete, it is inevitable that there will be occasional attempts to determine the Xserver socket while the Xserver is still starting and the Xserver socket is not yet present. In this case the variable "displayNumber" is deliberately set to '' to give it a defined value. This is then used again to deliberately try to connect to the Xserver socket "X", which triggers a new call to "getDisplayNumber". In this way, up to 100 attempts are made to determine the Xserver socket before it is assumed that the Xserver could not be started successfully and the error message "Connecting to the Xserver failed." is issued.

Once the Xserver socket used, and thus the display number, has been determined, it retains its validity until the connection to the Xserver is lost. Since the plugin listens to the Xserver socket, the described procedure is triggered again in this case in order to find out the new Xserver socket.

Until then, however, the known display number can be reused by the functions of the plugin via the variable "displayNumber" and there is no need to determine the display number again.

Ultimately, it's a matter of balancing either avoiding a global variable or function calls that cannot return anything other than the already known result at the time they occur.

Nonetheless I'll reconsider using the "getDisplayNumber" function instead of the "displayNumber" variable. The fact that the logs will from time to time show a failed connection attempt to a non-existent socket in the Xserver startup phase would not change though.

danowar2k, 2021-11-11:
I discovered this behaviour because I somehow configured my Volumio server in a way that during a running kiosk instance (where an X server should be running in the background) I tried to save some settings for the touch screen and it failed every time with the error message that the file /tmp/.x11-unix/X could not be found. So somehow there is a point during runtime where the displayNumber variable is set to an empty string, an X server is running at /tmp/.x11-unix/X0, but getDisplayNumber isn't called anymore.

If it was just a case of "I guess I'll have to wait until the displayNumber is detected correctly" I wouldn't mind...

danowar2k, 2021-11-11:
Background: I had some running cronjobs that

* shut down the kiosk at 9 PM, turning off the backlight
* started the kiosk at 7 AM, turning on the backlight

gvolt, 2021-11-11:
> ... I tried to save some settings for the touch screen and it failed every time with the error message that the file /tmp/.x11-unix/X could not be found. 

What settings did you try to save? Did you try saving them after your cron job had shut down the kiosk?

danowar2k, 2021-11-11:
> Did you try saving them after your cron job had shut down the kiosk?

That would be the easy way out, cause if the kiosk has shut down, the X server isn't running either, so no /tmp/.x11-unix/X0 either.

No, I remember trying to save the "seconds to wait until screensaver starts" setting a few times before 9 PM, so it was when the X server was running.

gvolt, 2021-11-11:
> No, I remember trying to save the "seconds to wait until screensaver starts" setting a few times before 9 PM, so it was when the X server was running.

Is this still happening, i.e. can you replicate the issue?