volumio/volumio-plugins #660: Volumio 3 - Spotify Connect2 Plugin - Fails to display Spoitify Account Configuration

FJRPilot, 2022-02-21:
After installing the Spotify Connect 2 Plugin and enabling it, the Spotify Connect Configuration page fails to display the user account and password used to log into Spotify with.

[image]
[image]

I have tried installing using the instructions on this Github Repository as well as downloading the plugin from the Plugins page (V3.0.8), both exhibit the same behavior

hendrix04, 2022-06-08:
Spotify Connect doesn't require credentials. You connect to it using spotify on your phone and it uses the credentials from your phone.

FJRPilot, 2022-06-09:
Thanks for following up on this Hendrix, I was confusing the Spotify plugin Vs the Spotify Connect plugin.  It’s all good now, or will be once the Spotify plugin get rewritten.  

Cheers!